LibreSolar/bms-c1 #1: Choosing internal power supply buck regulator

EarthLord, 2021-11-29:
Some options are 

- Rohm's [BD9G341](https://octopart.com/search?q=BD9G341&currency=USD&specs=0) or [BD9G500](https://octopart.com/search?q=BD9G500&currency=USD&specs=0)
- [MIC28512](https://www.mouser.in/c/?q=mic28512) available in Mouser. 2A output capability may not be enough if you want to support power for driving contactors and also a cellular modem. There are higher rating Microchip parts but they aren't available.

martinjaeger, 2021-11-29:
Hi Prithvi and thanks for your suggestions!

I'm currently considering the [TI LMR16020](https://www.ti.com/product/LMR16020), which is in stock at TI and at LCSC. There is also a 3A version (LMR16030) available.

Only problem: The 65V absolute maximum input voltage is a bit tight for 48V applications. There is also the [LMR38020](https://www.ti.com/product/LMR38020) with 80V nominal max, but it's not in stock anywhere. You can only get some pre-production samples from TI.

Compared to the proposed Rohm part, the LMR16020 has much better light load efficiency, which is quite important for this application, I think. The MIC28512-1 variant looks good as well in terms of light load efficiency. However, it has a quite unique footprint and I guess it would be easier to find a second source part for an SOIC-8 footprint (even though manufacturers seem to deliberately choose pin assignments that are incompatible to other parts, even e.g. within TI).

At the moment I'm not planning to add contactor drivers to the power PCB, as it will have the MOSFETs.

I was planning to set the power supply output voltage to 4V so that it can feed a GSM modem. The MCU and other parts of the circuit would then be powered by a low-current LDO providing 3.3V output. Do you think that makes sense?

EarthLord, 2021-11-29:
LMR38020 looks great with 45uA Iq. We had done a survey a few months ago and this wasn't available then. 

65V absolute limit is a no go when considering various battery pack possibilities, especially since 15s NMC packs can reach quite close to this limit. We got some 2nd life EV packs of this sort. 

Yes, we have a similar architecture of 4.2V and then an LDO for 3.3V. 4.2V because it also powers a charger IC for a backup battery. In case of this BMS, even 3.7 or 3.8V could work to reduce some loss from the LDO, depending on the dropout voltage of the LDO.

martinjaeger, 2021-11-29:
Good to know that you have 15s NMC packs. I wonder why they are 15s. Did you choose to put that many cells in series or did they come as a pre-assembled module with that many cells. As far as I know, most 48V batteries in automotive applications based on NMC cells only have 12s or 13s.

One option to be able to use a 60V-rated SMPS is to add a source follower between battery and input to reduce the voltage for such (I guess seldom) types of batteries. I've done that for the [MPPT 2420 HC](https://github.com/LibreSolar/mppt-2420-hc/) in order to allow power supply from up to 80V at the solar panel side. This will of course impact efficiency. It would be better to avoid this extra effort and have an SMPS that supports enough input voltage.

We could just try to get the pre-production sample of the LMR38020 from TI and hope that towards the end of next year it will be available from actual production. We can't really predict availability of parts anyway and whichever SMPS we choose, there will be some risk involved that it's not going to be available.

EarthLord, 2021-11-29:
The 15s pack was from the EV OEM. I guess they were aiming for 60V pack. 

Yes, looks like LMR38020 is the best bet at this time.