LibreSolar/bms-c1 #2: WSMS5515 as shunt resistor

EarthLord, 2021-11-29:
Two possible issue with this resistor 

- 5% tolerance, although this can be mitigated by storing a calibration value in the MCU on manufacturing
- Using this resistor as a connection point for BAT- and PACK- would put quite a bit of mechanical stress on the resistor with lugs connection with 25 mm2 wires.

What value of resistance is planned to be used and is 3W power dissipation with the resistor enough?

EarthLord, 2021-11-29:
An alternative could be to use Wurth connectors for BAT- and PACK- and use a 1% SMD shunt resistor such as this [8W, 1%, 0.5 mOhm SMD resistor](https://www.mouser.in/ProductDetail/KOA-Speer/PSL2NTEBL500F?qs=5aG0NVq1C4yLtsrj1PD%252BsQ%3D%3D) with a layout like as described in this [article](http://www.analog.com/media/en/analog-dialogue/volume-46/number-2/articles/optimize-high-current-sensing-accuracy.pdf) by ADI to implement Kelvin connection.

martinjaeger, 2021-11-29:
Yep, 5% tolerance is indeed not ideal, but it could be calibrated in firmware during end-of-line check to increase accuracy.

I do agree that the connection could create stress. We could try to support the shunt using the plastic enclosure.

Based on maximum heat dissipation in the shunt (3W), the ideal value would be 300 µOhm (for 100A max. continuous current). However, this variant is currently not in stock anywhere. Mouser only has the 100 µOhm type. I've still got to check the resulting resolution for the bq76952.

The alternative SMD resistor you suggested looks good. I've used similar parts from Bourns in the [BMS 8s50 IC](https://github.com/LibreSolar/bms-8s50-ic). However, I'm a bit concerned with the 100A current pushed through one rather small SMD soldering connection. Also 8W heat dissipation seems quite a lot for such a small component without attaching it directly to a heat sink.

Many other BMSs use multiple SMD resistors in parallel, see e.g. this [Pylontech teardown](https://powerforum.co.za/topic/8524-pylontech-us2000-plus-teardown/).

Do you have experience with current in the 100A range on "normal" PCBs? Which copper thickness would you think is required for the PCB?

Thanks also for the link to the document regarding kelvin connection. I've always connected from below in the center of the resistor, but if the other connection is more accurate I'm happy to follow the paper. Might get tricky if we have multiple resistors in parallel, though.

EarthLord, 2021-11-30:
The manufacturers of the SMD resistors seem pretty confident of their capability to handle the power dissipated. For example [PSJ2NTEBL500F](https://www.digikey.in/en/products/detail/koa-speer-electronics-inc/PSJ2NTEBL500F/10396748) says the thermal resistance is 6.7 °C/W, so with say a bad ambient temperature of 50°C, a 100 A current through this will get it to 83.5°C, while still staying within the rated power. 

Yes, I saw the Pylontech teardown recently too. I'm wondering how the accuracy of such a system is. The ADI article says how the soldering is done will influence the resistance quite a bit, although considering stencil and reflow soldering should give uniform current split among the resistors. The advantage of this is possible lower cost of say 8 quantity of 1%, 4 mOhm resistors than 1 of 0.5 mOhm 1% resistor of higher wattage.

The [tool to graph the error](https://www.ti.com/product/INA228#order-quality) of INA228 is interesting, if you've not seen it before. I guess something similar is also valid for bq76952 and this is important for Coulomb Counting.

We have not developed any circuitry with currents of magnitude of 100A. I'll get back here with results of testing another BMS's PDU of 2 Oz Cu and 75A continuous current. 

Also take a look at LiBal's [iBMS](https://ehv.mydigitalpublication.co.uk/publication/?i=699826&article_id=3974942&view=articleBrowser&ver=html5). Its crazy what they have achieved in a 65mmx200mm board. Its capable of 100A continuous current and 200 mA balancing current.

martinjaeger, 2021-11-30:
The LiBal BMS is indeed very cool. Looks like the MOSFETs are on the bottom. I wonder how they can be attached to a heat sink.
Unfortunately, there is not much further information available. They'll probably have MOSFETs + current sensing in series on the high-side and GND is connected via one of the small connectors. Saves two high-power connections, but is not possible with bq76952. I wonder what the huge hole in the PCB on the right is for...

EarthLord, 2021-12-03:
Here's a result from a test where at the end of 12 min test the recorded temp is 80C (Flir camera shows 5C more than a Fluke thermal gun). The conditions are 

- 30C ambient temperature
- 78A charging current
- IPB048N15N5ATMA1 MOSFET with Rds on max of about 5 mOhm
- Test time of 12 min
- 2 Oz Cu board

[image]
[image]

We'll do a longer test and get back.

martinjaeger, 2021-12-03:
Thanks for uploading the test results. So the heat dissipation should be around 5 Ohms / 6 * 2 * (80A)^2 = 10.6W, which is quite similar to the configuration mentioned in #3 at 100A. So it looks promising in general. Did you have any heat sink / spreader attached at the bottom or was it just in free air?

EarthLord, 2021-12-04:
[image]

There are two bus bars at the bottom and it was in free air.

martinjaeger, 2023-04-15:
The chosen SMD shunt performs well for 100A, see test report.

closed 2023-04-15 by martinjaeger