LibreSolar/bms-c1 #3: Necessity for two WP-BUCF connectors in parallel?

EarthLord, 2021-11-29:
Two Würth WP-BUCF power connectors are planned for Bat+ and Pack+ each. The possible issues with this are

- majorly the increase in cost for 2x of these costly connectors
- Lugs are designed in such a way that the bottom surface is made completely flat and thus provide max contact with terminal block. The 180 degree flip of one lug may not provide enough contact.
- Also depending on torquing of the screw, the resistance of connection would vary, thus not splitting the current in half.

An alternative would be to use one of [7461383](https://octopart.com/search?q=7461383&currency=USD&specs=0) only such that it can handle all the current with its greater mating points with PCB and it is also sightly less expensive. For applications with vibrations it'll also be easier to apply thread lock glue. 

If some space can be made at the corner of the PCB by moving the RJ45 connectors, M6 part also could be an option.

EarthLord, 2021-12-01:
[image]
A possible option would be to use parts from [Lugs Direct](https://lugsdirect.com/PCBsolderable-lugs.htm). The wires would have to use [these kind of termination](https://www.swaonline.co.uk/cable-terminations/copper-pin-terminals/uninsulated-reducing-pin) in order to use them, bare wires get corroded and increase resistance. 
[image]

In an upcoming BMS from another vendor we'll be using [B2C-PCB-PT-HEX](https://lugsdirect.com/heavy-duty-pcb-terminals-pt-pcb.html) (shown in pic above). These particular connectors are used in APC inverters. It is rated to 115A and there are other connectors with higher rating with the same company. 

As for as the connector in your screenshot, the max I've seen on [Aliexpress](https://www.aliexpress.com/item/32758412452.html) is M5 screw based that is supposedly rated upto 100A. 

Yes, it makes assembly simpler with soldering than press fit, which is important for an open source project.

martinjaeger, 2021-12-01:
I'm currently checking out a new approach which avoids the expensive terminals completely. The idea is to use a custom-made (yet simple) copper bus bar of e.g. 2 mm thickness with a clinch nut pressed in from the back. The bus bar would be soldered directly onto the PCB and can spread the current as well as heat.

[image]

[image]

By using only SMD parts, we have a plain surface at the bottom which can be easily attached to a heat sink. It would need some milling to provide space for the clinch nuts, though. Maybe even a thick aluminum heat spreader is sufficient for cooling purposes, if e.g. attached to a housing outside wall.

This suggestion addresses some of the other points raised by @EarthLord aswell.

Ideal MOSFET in terms of heat dissipation would be [IPT015N10N5](https://www.infineon.com/dgdl/Infineon-IPT015N10N5-DS-v02_02-EN.pdf?fileId=5546d4624a75e5f1014ac94680661aff), where 4 in parallel should be sufficient. Rds(on) at 80°C is max. 2 mOhm, leading to a total of 1 mOhm (4p2s) and 10W of heat dissipation at 100A.

There are cheaper alternatives with higher Rds(on), also from other manufacturers.

EarthLord, 2021-12-02:
Looks great in many aspects - components assembly, connection of lugs, lowering resistance between MOSFETs and connectors, lower PCB copper thickness requirement and aesthetics. If needed the length of the board can be increased to have either 1. more spacing between the MOSFETs for better thermal performance or 2. make it 5p2s with maybe some DNP if its a lower rated requirement. H1, H2, H3 holes for attaching heat sink is also a good idea! 

Feedback from our mechanical engineer - thing you've to keep in mind is not to use steel clinch nut with copper. They have different hardness and thermal expansion. So steel clinch nut won't be secure for longtime over multiple thermal cycling. Copper clinch nut on the other hand would be good, if they can be found.

martinjaeger, 2021-12-06:
> Feedback from our mechanical engineer - thing you've to keep in mind is not to use steel clinch nut with copper. They have different hardness and thermal expansion. So steel clinch nut won't be [...]

I talked to [KVT](https://www.kvt-fastening.de/en/products/brands/pem/) about above point and they don't see any issue with different hardness and thermal expansion. The stainless steel nut (CLS variant) is suitable for copper and aluminum material. It is used in multiple different high-current applications. Also the old Tesla model S battery (and potentially newer ones as well, but I don't know) uses a clinched nut in a copper shunt.

photon-delight, 2022-02-19:
> The bus bar would be soldered directly onto the PCB and can spread the current as well as heat

How would these bus bars be soldered?

martinjaeger, 2022-02-25:
> How would these bus bars be soldered?

The idea is to solder them together with other components during the reflow process.

I have changed the design of the busbars to have press-fit studs instead of press-fit nuts because it avoids milling of the heat sink and it should also be easier for assembly because the cable lugs are kept in place already without the bolt. You can find an updated image in the main Readme of the repository.

closed 2023-04-15 by martinjaeger