LibreSolar/bms-c1 #4: Plan for board to board connections

EarthLord, 2021-12-07:
What would be the plan for board to board connections? As I can see one would be for BMS <-> PDU and other would be for BMS <-> wireless communication board.
Is the plan to use direct board to board connection or with a cable, say FRC cable in between?

martinjaeger, 2021-12-07:
That's exactly what I've started to work on yesterday. I'm currently thinking of a simple pin header between power board and control board.
[image]

For connection of the comms board I'd suggest to use the UEXT connector, which works both with ribbon cables or a direct pin-header connection. A GSM board would need an additional power connector (e.g. JST XH or PH) as the ribbon cables are too small for 2A over a longer distance.

I'm also considering to remove the RJ45 connectors from the board and route CANH and CANL into another MicroFit type connector. The RJ45 jacks should be accessible from the outside of the battery housing. Having them directly at the board of the BMS would restrict the design of the battery housing quite a lot.

What do you think?

EarthLord, 2021-12-09:
For UEXT connector board to board mounting, you are planning to use generic 5x2 female pin header?

Considering that UEXT standard specifies 3.3V for power, 4V can't be used for GSM communication anyway too. Like you said, a thicker wired 2 pin connector for 4V can be used present in case a module needs it. 

With respect to BMS <-> PDU, would it be good to also have the possibility to have an option to mount the two separately, not one on top of other? Especially considering the PDU can heat up and this can affect the BMS in terms of accuracy and balancing current. 
For this maybe a UEXT connector model would work for BMS <-> PDU?

With respect to CAN, yes a MicroFit type of connector makes more sense for the reason you mentioned.

martinjaeger, 2022-02-25:
I've had lots of back- and forth thinking for the connection between the control PCB and the power PCB and finally decided to go with two rather simple 1x6 pin sockets/headers with 2mm pin pitch.

Control PCB:

[image]
[image]

Power PCB:

[image]

## Reasons

- Cheap solution compared to other board-to-board connectors
- Stacking height is lower than with 2.54 mm pitch connectors (approx. 8 mm total)
- There was no space in the middle of the control PCB because of balancing resistors, etc.

Unfortuntately the connectors on the top PCB require an additional THT soldering process, but I did not find any viable solution for reflow soldering on top side. Bottom-entry sockets [like this one](https://www.we-online.com/katalog/de/PHD_2_54_SMT_DUAL_SOCKET_HEADER_BOTTOM_ENTRY_6100XX243021) are very expensive and require longer than normal pins at the bottom side.

BTW: All other parts on the top PCB are SMT. Only the optional UEXT header and the additional 5V aux. power supply connector are THT, but they will not be needed in many applications. The programming connector is press-fit (will post another issue with more details).

## Considered alternatives

- Flat ribbon cable -> not ideal from EMC perspective as the current sense connection should be as short and close to the BMS ASIC as possible.
- FPC -> similar issues as with flat ribbon cables
- Other board-to-board connectors: Usually very short distance between PCBs so that the MOSFETs and busbars of the bottom PCB don't fit anymore.

EarthLord, 2022-02-28:
Lots of valid points, makes sense to have this solution that you've proposed. Would it be better to have the two connectors next to each other so that FRC is also an option when developing/debugging when both boards need to be accessible? 
Which means it'll be a dual row connector on both the sides and it'll be 2.54 mm pitch which increases the height too. 
The other option is building a custom cable for the two 1x6 connections when debugging if needed when the two boards can't be stacked.

photon-delight, 2022-02-28:
I'm a bit concerned about the strength of these SIL SMD headers on the power PCB.  I would be more inclined to use  DIL SMD headers as they would be a bit more resilient against lateral force, and there seems to be enough space on the power PCB.  I understand that the two boards are mechanically secured using the 3 standoffs, but it would reduce the fragility of this PCB before being mechanically secured.

b0661, 2022-02-28:
Some thoughts about production when looking at the pictured boards: 

- Usually you take the "long" side of a board to support in pick & place and soldering. One "long" side is only half, which is not optimal for mechanical support during production. 
- Some of the components are very close to the board border which increases the risc of breakage during production. Also if you panelize several boards this is a risk of failure due to mechanical stress when cutting the boards . 
- Are the board cutouts really necessary? Boards without cutouts are usually cheaper. 
- There is only one fiducial in the picture, at the edge of the board (None on the power board?). Fiducials should (also) be placed near to the high density components. There is a higher risk of misplacement by one pin.

martinjaeger, 2022-02-28:
Thanks for all your feedback.

@photon-delight Would you suggest two 2x3 pin headers instead of the two 1x6 pin headers? Wouldn't they also have 6 legs soldered to the PCB, so the maximum lateral force would be the same?

@b0661 

> * Usually you take the "long" side of a board to support in pick & place and soldering. One "long" side is only half, which is not optimal for mechanical support during production.

What do you mean exactly by long side? Do you mean that the ratio between long and short side is high for the power PCB so that it could bend?

> * Some of the components are very close to the board border which increases the risc of breakage during production. Also if you panelize several boards this is a risk of failure due to mechanical stress when cutting the boards .

Good point, the one large MLCC is really close to the edge. I'll try to move it further inside. In general the space was quite limited in some locations so that I couldn't keep much space at the edges.

> * Are the board cutouts really necessary? Boards without cutouts are usually cheaper.

Yes, I think they are necessary. The board height is limited to 65 mm so that it can fit at the end of a pack with typical 18650 Li-ion cells. Thats also why all connectors have to go out at the short sides of the PCB. The ESP32-C3 module antenna is best placed at the board edge, so we need to cut out the board in order to stay below the maximum height of the assembly. The smaller cut-outs next to the cell connector are necessary for the tool of the nuts for the cable lug.

> * There is only one fiducial in the picture, at the edge of the board (None on the power board?). Fiducials should (also) be placed near to the high density components. There is a higher risk of misplacement by one pin.

There are 2 fiducials on each board, but they don't have a `FID` marking. I actually forgot to make the `FID1` marking invisible because I don't think they need an identifier.

martinjaeger, 2022-02-28:
> Lots of valid points, makes sense to have this solution that you've proposed. Would it be better to have the two connectors next to each other so that FRC is also an option when developing/debugging [...]

Probably we'd have to build a custom cable for debugging, as also the signals in the top and bottom connector are chosen such that they are close to their source/sink (e.g. discharge MOSFET side or charge MOSFET side), so they don't have to be routed across the entire board.

Light411, 2022-03-04:
Hey, 

I do share concerns with what have been said above regarding the pin headers. 
Have you considered pogo pins ? It has some perks, as it won't suffer from lateral forces and will have less chances to get teared out on the long term. 
This is what I used to replace most of my headers in mezzanines and I'm really have with it. You basically separate the mechanical constraints from the electrical contact which is great.
https://www.mouser.fr/ProductDetail/Mill-Max/0906-0-15-20-76-14-11-0?qs=DGF5YW3GH8vs5H4c8CyXhQ%3D%3D

martinjaeger, 2022-03-11:
I have now updated the footprint such that the pad is below both sides of the connector. This allows to use parts with pin 1 left or right and in addition to that increases rigidity a bit as the pad size is bigger.

[image]

I'll try this out for the first batch and if it is not sufficiently stable we can use the 2x6 connector instead.

The pogo pins as suggested by @Light411 are quite expensive IMO. Also they don't seem long enough to cover the required distance between the board, so I'll keep that in mind as a backup solution if the current solution does not work well.

martinjaeger, 2023-04-15:
Board-to-board connections are not required in most recent design anymore.

closed 2023-04-15 by martinjaeger