LibreSolar/bms-c1 #5: Footprints for thermal jumpers?

EarthLord, 2022-01-05:
Add footprints for thermal jumpers such as 0612 variants of [THJP series by Vishay](https://www.vishay.com/docs/60157/thjp.pdf)? The footprints can be from the common drain points of the MOSFETs and send heat to their source planes as well. They can be populated if necessary for additional heat sinking.
This [video by EEVBlog](https://www.youtube.com/watch?v=q3RhWuXFixU&t=782s) was where I got my insight into this interesting part.

martinjaeger, 2022-02-25:
Hi @EarthLord. Thanks for the suggestion.

I've just uploaded some thermal calculations for the heat sink. At least for the MOSFETs the transfer from the MOSFETs to the PCB and to the heat sink is not an issue. The thermal bottleneck is more the heat transfer from the heat sink ambient. So for applications with 100A requirement, the BMS will have to be thermally attached to the battery housing to transfer the heat out of the enclosure.

The shunt resistor is more critical in terms of temperature increase (as you also mentioned via e-mail). However, I hope that the bus bars will act as heat spreaders and improve the heat transfer towards the heat sink. As this is 3-dimensional heat transfer, it is more difficult to estimate, so I will have to do some testing.

EarthLord, 2022-02-25:
Okay yes, looks like shunt will get more toasty. Looking at the power derating curve of the [shunt resistor](https://www.koaspeer.com/pdfs/PSJ2.pdf), it looks like at higher current the temp increases which further decreases the power rating. 

[image]

How about a shunt like one in this pic from Seplos BMS? I don't have any part numbers for this yet.

EarthLord, 2022-02-28:
I am unable to get the part number for shunt like the one in the pic above, which seems apt as it'll remove the need for bus bars as well for the -ve terminals as well as have better thermal performance. Its from [Seplos BMS](https://www.seplos.com/sale-13707067-can-bus-rs485-48v-200a-5-500ah-8s-16s-lithium-iron-phosphate-lifepo4-bms-monitoring-system-battery-m.html) where they are using this shunt for up to 200A. 
It shouldn't be too expensive too since they are selling a BMS at 125 USD at single quantity.

martinjaeger, 2022-02-28:
The thermal performance could even be compared without a board change, as the WSBS8518 needs to be mounted somewhere outside the board anyway. It exceeds the maximum board height of 65 mm defined by the 18650 cells. The one in the Seplos BMS looks shorter, but I couldn't find any comparably small shunt resistor either during my previous research.

The problem with the solution with these large shunt resistors is the additional cost for the screw terminals on the PCB and that the Kelvin connection to the board is somewhat tricky.

Maybe we just need to go for 0.3 mOhm if the 0.5 mOhm SMD resistor gets too hot.

martinjaeger, 2022-03-10:
I won't add the thermal jumpers or the shunt resistor for the first batch, but we'll keep this open in case testing shows that performance of current design is not sufficient.

martinjaeger, 2023-11-02:
Testing showed that 100A work well without thermal jumpers.

closed 2023-11-02 by martinjaeger