LibreSolar/bms-c1 #7: Choice of ESP32 variant

EarthLord, 2022-02-25:
C3 Mini module of ESP32 is chosen. Any reason for C3 vs say S2?

martinjaeger, 2022-02-25:
It's mainly due to space constraints. The "larger" ESPs are not available with a small footprint module. And compared to the STM32G0B1 the ESP32-C3 is still more powerful.

Would you have any particular use case which requires an ESP32-S2?

martinjaeger, 2022-02-25:
I just see that the S2 doesn't even have Bluetooth, which is the main reason for me to have ESP32 support.

Also I kind of like that the ESP32-C3 has a RISC-V core :)

EarthLord, 2022-02-25:
Sounds good. Just asked to understand your decision process. Yes, was quite a pleasant surprise when ESP32 C3 came out with a RISC-V ISA based core.

closed 2022-02-28 by EarthLord