LibreSolar/bms-c1 #10: Creepage distance Battery voltage

JochenKr, 2022-03-02:
The creepage distance for the battery voltage is only 0.3mm to GND.
To reduce the risk of arcing, it makes sense to increase the distance to at least 0.7mm in respect to IPC9592.

example:
[image]

martinjaeger, 2022-03-10:
Swapped the pins and increased the GND plane gap on top layer from 0.3 mm to 0.5 mm:

[image]

[image]

You suggested 0.7 mm, which would make quite a few GND planes on the top layer disappear. I'm wondering if 0.7 mm is only required for dirty / moisty environment, as it is violated by the IC anyways:

[image]

PDSG pin is pulled up to the battery voltage during normal operation and it's right next to a GND pin:

[image]

martinjaeger, 2022-03-15:
The influence of the impedance makes sense.

I've swapped the pins as suggested and increased the gaps to 0.5 mm for now. Leaving this open so that future revisions can be further improved.