LibreSolar/bms-c1 #11: Ground on BQ76952

JochenKr, 2022-03-02:
The ground loop around the IC can be improved.
[image]

the best would be to have a solid GND plane below the IC. This seems not to be easy.
An improvement could be to put TS3 signal to bottom layer. By this there is a solid connection of GND on top layer possible,
By slightly moving RST_SHUT and TEMP4 it is possible to add two additional vias in that area to connect the planes.
[image]

martinjaeger, 2022-03-03:
Good spot. I also realized that the routing around the IC with the many traces in the bottom layer is not ideal. But also didn't want to go for a 4-layer PCB just because of that. I'll try to move things around a bit so that the top plane bridges the gaps created by the traces in the bottom plane better.

martinjaeger, 2022-03-10:
After watching [this amazing video](https://www.youtube.com/watch?v=ySuUZEjARPY&ab_channel=Altium) as recommended by @Light411 I finally decided to switch to a 4-layer PCB. I thought I'd get away with the cheaper 2-layer version, but routing around the IC proves to be really difficult. I guess the small cost increase will be worth the improved performance.

Light411, 2022-03-11:
Super happy that it was helpful ! I'm sure it will greatly improve the ground loops issues. 
What layer stack-up will you choose ? Will you do 

- Signal 
- Ground 
- Ground 
- Signal 

as suggested by Rick, with a routed VCC ?

martinjaeger, 2022-03-11:
Yes, exactly. I was previously mostly using:

- Signal
- Ground
- VCC (3V3)
- Signal

But after watching his video it seems to make more sense to have two GND layers and a routed VCC. Maybe the VCC can be partly flooded on the bottom layer, otherwise there would be very few traces only.

commit 024bcb44 (2022-03-15): Updates based on review comments (v0.2)

This is the design revision that was ordered for the first prototype build.

The pinout of the cell connector and the board-to-board connector changed
compared to v0.1, hence it got a minor revision update.

The control board is now a 4-layer board to improve the grounding of the
PCB. Various other suggested improvements from the design review are also
included in this update.

Fixes #11, #12, #13, #17, #18, #19
closed 2022-03-15 by martinjaeger