LibreSolar/bms-c1 #9: 5V Buck converter

JochenKr, 2022-03-02:
Noise on the Vcc of 5V supplied devices should be better if the GND is connected on the output caps of the buck converter.
An additional 0603 100nF Cap on the output side could improve noise on the 5V.

[image]

EarthLord, 2022-03-03:
Yes, also for all decoupling caps it'll be good to have vias close to its GND pad to reduce inductance. Page 29 of this slide has the recommendations https://web.mst.edu/~jfan/slides/Archambeault1.pdf

Its great to see the tight input-output and diode-output loops of the buck regulator :)

martinjaeger, 2022-03-03:
For the connection to GND I was roughly following the TI recommendations in the [LMR16030 data sheet](https://www.ti.com/lit/ds/symlink/lmr16030.pdf?ts=1646304542517&ref_url=https%253A%252F%252Fwww.ti.com%252Fproduct%252FLMR16030):

[image]

In the datasheet of the LMR33630 it is even made more clear that they suggest to place the ground connection close to the IC and not close to the caps:

[image]

I've had devices with the LMR33630 in the EMC lab and they were perfectly fine, so I took a similar approach here.

In my understanding it is most important to have a **single** connection between the power supply GND and and the GND plane of the circuit. We could probably also place it close to the capacitors, but in that case we'd have to cut the plane on the top layer close to the U2 marking in order to avoid two different paths.

For the output side a smaller cap for high-frequency noise is probably not as important as for the input side, but looks like there is enough space, so I'll add one.

JochenKr, 2022-03-03:
Probably they did it to cool the IC?

It's definitely not urgent, especially as the currents will be very low. 
I agree that the best solution should be to have one single GND connection.