LibreSolar/bms-c1 #13: Thermal performance of balancing bleed resistors

EarthLord, 2022-03-03:
BQ76952 has 100 mA balancing current, relatively less. In any case, since the control board is small, just wanted to check if any rough thermal analysis was done on this matter?

martinjaeger, 2022-03-03:
Good point. I did thermal camera shots with the bms-8s50-ic in the past and the temperature increase was very low, something around 40°C at 20°C ambient. But it was a 4-layer PCB and the resistor value was 33R, so the situation will be a bit different here. I'll double-check and potentially try to add a bit more copper below the resistors.

martinjaeger, 2022-03-10:
As mentioned in the other PR, I'm switching to 4-layer PCB. In addition to that I added large pads on the top and bottom layers connected through thermal vias to spread heat, see below:

[image]

The picture doesn't show the two inner layers which provide another copper pour for heat spreading. This is very similar to previous designs and I'm confident it will provide enough heat removal.

martinjaeger, 2022-03-15:
Closing this now as I don't think there will be any issue with the updated design anymore. If testing shows different results we can re-open the issue.

closed 2022-03-15 by martinjaeger