LibreSolar/bms-c1 #14: Voltage rating of D1 and D2 in bms-power-100a

EarthLord, 2022-03-04:
Considering we can use 16s NMC cells too, it would be better to get the 64V or next higher Reverse Stand off Voltage Vr rated SMC TVS diode in bms-power-100a boards for D1 and D2.

martinjaeger, 2022-03-10:
Currently we cannot use 16s NMC cells because of the limited voltage range of the power supply.

SMCJ70A have a maximum clamping voltage of 113V, which is beyond the rating of the MOSFETs. So we'd probably have to choose different TVS diodes and MOSFETs depending on the maximum number of expected cells. For usual 48V applications with 60V max the SMCJ60A are the right choice, I think. Do you agree?

EarthLord, 2022-03-11:
Oh, just saw that LMR16030 is only up to 60V. Would it be good to limit the number of cells usable with NMC just because of the power supply?
We have got supplies of 15s NMC packs, not 16s as of now. So it would be good if at least 15s NMC is supported. Unfortunately as you mentioned earlier, these SOIC parts aren't pin compatible.

For power supply some options that I can think of are 

- Move to LMR38010 which is available and pin compatible with LMR38020. With this approach we can 
  - support higher cell count with NMC
  - replace with LMR38020 when it becomes available and then GSM can be supported as well. Before that we can figure out external supply for development
  - TVS can be moved to SMCJ70A to support higher number of cells. With max clamping of higher voltage, yes the protection is reduced. For small power spikes, this 1.5kW TVS can work too, although there are 5kW and 8kW TVS in the same package too, which will offer better clamping if the developer thinks the application would experience higher spikes.
  - MOSFET could be 100V Vds for now I guess.
- Other option for higher input voltage options like we discussed in this https://github.com/LibreSolar/bms-16s100-sc/issues/1 What about the Rohm parts in there?
- Or LMR16030 for the first version and then hope there's better alternatives for the 2nd version so that higher num of cells can be catered to. Only thing is that it'll be an untested part of the 2nd version.

martinjaeger, 2022-03-11:
I'm afraid that the only viable option right now is to wait for the LMR38020 to become available and till then use the LMR16030.

- The LMR38010 seems to be a very new part (didn't know it before) and it is only available in TI stock, but for some reason I'm not able to order at TI, at least it was not possible in the past. They seem to ship only to distributors or US companies.
- As discussed before, the ROHM parts have bad light load efficiency, so they are not suitable for a BMS.