LibreSolar/bms-c1 #15: CAN bus termination R & C with DNP

EarthLord, 2022-03-04:
Can a split CAN termination be added with DNP marking so that there is an option to add this if needed

martinjaeger, 2022-03-10:
As far as I know, there is no SOIC-8 transceiver with standby pin, 3.3V I/O and split termination due to lack of pins.

Apart from that, I think it is necessary to leave termination to the application where the BMS is embedded into. If we add the termination to the PCB, how can we have multiple BMSs in parallel? Would we need to have different PCB versions with and without termination so that the ones in the middle of a bus would not have the termination resistors?

Victron and several other manufacturers use termination connectors which can be plugged into the first and last device of the daisy chain. Maybe these connectors can also have a basic form of split termination if they have a GND pin.

EarthLord, 2022-03-10:
[image] 
Pic from this [TI article](https://e2e.ti.com/blogs_/b/industrial_strength/posts/the-importance-of-termination-networks-in-can-transceivers)
For split termination just GND of CAN transceiver is needed, right? Why would standby pin, 3.3V I/O be needed?

Yes, its a valid point that when multiple BMS are used or when other CAN devices are used, can termination can be designed according to the application. In any case if the footprint is there in the PCB, since it is an open source board if the developer wants, there is an option to populate the termination resistors. It adds to the flexibility the board offers.

In your schematic it is a Do Not Populate (DNP) part which is left open by default.

martinjaeger, 2022-03-11:
I meant that e.g. the TJA1042 transceiver we are using is available in a special variant for split termination, see [datasheet](https://www.nxp.com/docs/en/data-sheet/TJA1042.pdf), which stabilizes the voltage at 0.5 * VCC. However, the variant with split termination is only available for 5V I/O. If you want that spit pin + 3.3V I/O + standby, a larger package with >8 pins is needed.

Probably the split termination without the dedicated split pin is still better than standard termination, so we can consider it.

The space around the CAN transceiver and connector is pretty packed. I'll see if I find some space and if yes I can add the parts as DNP (as suggested).

EarthLord, 2022-03-11:
Thanks Martin. Didn't know of the Split potential output feature of transceivers, good to learn.

martinjaeger, 2022-03-15:
I didn't find the space for the 3 additional components without lots of shifting of other parts, so I'll leave this enhancement for a next revision.

martinjaeger, 2022-08-27:
Split termination was added in new design:

[image]

closed 2022-08-27 by martinjaeger