LibreSolar/bms-c1 #27: Thermal performance of N-MOS switches

EarthLord, 2022-06-22:
Hello, just curious to see if any thermal performance numbers have come up from any initial tests.

EarthLord, 2022-07-06:
https://lcsc.com/product-detail/MOSFETs_HUAYI-HYG022N10NS1TA_C2925050.html might be an alternative that can be used in the interim.

martinjaeger, 2022-08-02:
Those HUAYI MOSFETs are indeed quite interesting. There is even one with almost the same specs as the originally planned IPT015N10N5: https://lcsc.com/product-detail/MOSFETs_HUAYI-HYG015N10NS1TA_C2979263.html

I did some research about the quality of that manufacturer and found that they actually produce for Infineon, so those MOSFET may be of good quality, but lower price: https://www.mouser.com/PCN/Infineon_20176A.pdf

@EarthLord do you have any experience with them?

EarthLord, 2022-08-04:
No, don't have experience with this manufacturer yet. I found them while looking for TOLL8 package mosfets during this shortage season. Maybe good to try them out based on your finding that they manufacture for Infineon.

martinjaeger, 2023-04-15:
Final test report including thermal performance tests was added to the repo:
https://github.com/LibreSolar/bms-c1/blob/main/testing/v0.3/README.md

closed 2023-04-15 by martinjaeger