LibreSolar/bms-c1 #28: Current shunt footprint recommendation

EarthLord, 2022-06-22:
https://www.analog.com/media/en/analog-dialogue/volume-46/number-2/articles/optimize-high-current-sensing-accuracy.pdf This article of ADI provides some recommendation for a 4 point footprint for shunt. Would something like this be done in the next version?

Also based on thermal performance, a 5931 package SMD shunt might be better than a 3920 one?

martinjaeger, 2022-07-05:
Yes, taking the larger footprint makes a lot of sense and will be implemented in the next revision. The cost increase is quite low.

I will also update the sense traces, but probably not exactly like in the paper you linked, which is designed for much lower current. I don't want to waste any trace width in the high current path with the via outside the shunt, so I would place the traces towards the center of the shunt.

martinjaeger, 2022-08-27:
Implemented in newest revision:

[image]

closed 2022-08-27 by martinjaeger