LibreSolar/bms-c1 #29: Charge pump capacitor value estimation

B-hafedh, 2022-06-28:
[image]

How do you estimate the charge pump capacitor value ?
 4*16nF*10=680nF

B-hafedh, 2022-06-28:
> 
Thank you,
but there is no information about it in bq76952 datasheet, can you provide me the document please.
and i have another question: what is the role of the schottky diode & zener diode connected to VC0, i thought they are used for reverse polarity protection, but you already used a schottky diode connected to VC16.

martinjaeger, 2022-07-01:
It was not necessarily the datasheet. There are lots of documents for this chip provided by TI, I don't remember which one it was.

In any case you can also estimate the value yourself. It is a trade-off between startup-time (faster if capacitance is lower) and voltage drop during switch-on event (lower if capacitance is higher).