LibreSolar/bms-c1 #31: BAT+, CHG+ and PACK+ voltage measurment

EarthLord, 2022-07-03:
I just realized that there's no provision for BAT+, CHG+ and PACK+ voltage measurement or did I miss it in some corner of the schematic? Or can BAT+ be measured independently by the BQ chipset?

martinjaeger, 2022-07-05:
Yes, you're right, it was missing. I've already included a PACK+ measurement in the next revision of the board (to be finished and published in a few weeks).

For bus pre-charge I think PACK+ is sufficient. What do you think CHG+ voltage measurement would be useful for?

EarthLord, 2022-07-05:
CHG+ is useful for triggering some alarm by detecting if there is an MOSFET issue of short circuit by measuring CHR+ when MOSFET.

martinjaeger, 2022-07-05:
Not sure if I understand what you mean. Could you elaborate a bit more?

I'd like to avoid a resistor divider at the CHG+ because it would continue draining the battery through the CHG MOSFET body diode even if the battery is in deep sleep.

EarthLord, 2022-07-06:
Sorry for the bad explanation. MOSFETs when stressed with over current go bad by short circuiting without gate control - but in our case the short circuit will have higher resistance than when turned on when operational. My intention was to be able to detect this situation. I forgot when I posted this that the MOSFET configuration is with the drains tied together. With source tied together its easier by checking the common source voltage when both MOSFETs are turned off.

With drain tied together it might be possible by accounting for the diode forward drop. Which ADC do you want to use for BAT+ and PACK+ measurement?

martinjaeger, 2022-08-27:
The bq does actually have its internal BAT+ and PACK+ measurements, so I decided to not use the MCUs ADC for this sort of measurement, as the voltage divider would also result in an additional continuous (very small) current flow. In addition to that, the ADC of Espressif chips is usually not very good...

closed 2022-08-27 by martinjaeger