LibreSolar/bms-c1 #39: Intention of CHG+ pad

nik21-nik, 2023-03-18:
Hi,

I have already checked the manual (I hope I have not overlooked anything, then sorry for that). I am not sure what the intention of the CHG+ pad in the PCB layout is. Obviously the port powers the I would call it "Predischarge circuit". 
Is the exposed for the charger connection? Then why would I solder a cable onto a pad instead of using the power port connector like you did with BAT+, BAT-, PACK+ and PACK-.

Thank you!

martinjaeger, 2023-03-18:
The CHG+ pad is for experimenting. That's why it only got a zero-cost pad instead of a proper terminal.

Some users were suggesting to have separate charge and discharge paths as seen in many e-bike BMSs. However, in e-bikes these paths are in parallel, such that accidental overcharging through the discharge path cannot be interrupted. This was not an option for our BMS for safety reasons.

Using the CHG+ pad for chargers slightly increases the efficiency, as the heat dissipation in the DIS MOSFET can be avoided. In addition to that, a charger may be able to detect the battery even if it is in deep sleep mode.

nik21-nik, 2023-03-20:
Thank you very much for the explanation.

closed 2023-03-20 by nik21-nik