LibreSolar/bms-c1 #40: D16 Schottky diode inverted?

LorenzoGuerro, 2023-04-25:
Hello,

I simulated the reverse polarity protection circuit and it somehow doesn't work. There is a whole amount of current flowing over the DSG pin through the Schottky diode D16 with a reversed charger. A Texas Instrument application document shows this diode inverted. Is not the R51 68k resistor meant to drive the gates and the (then inverted) diode to turn off the discharge MOSFETs faster?

Then, if there is a reversed charger connected, voltage drops over the R51 68k resistor.

martinjaeger, 2023-04-28:
Hi Lorenzo.

Thanks for reporting your findings.

As far as I can see, the [TI reference design](https://www.ti.com/tool/TIDA-010208) has the diode D37 in the same direction.

[image]

That being said, I haven't tested the reverse polarity circuit in real life yet, as I didn't have too many boards yet to potentially destroy them. I'll try to test it with one of the early prototypes by slowly increasing the reversed voltage with a current-limited power supply to see what happens.

Which software did you use for simulation of the circuit. Would you mind sharing the file if it's a freely available software like LTSpice?

LorenzoGuerro, 2023-05-02:
Please find here the simulation I made in Falstad Circuit Simulator: [simulation.zip](https://github.com/LibreSolar/bms-c1/files/11370771/simulation.zip)

Things I am unsure about:
The DSG pin outputs about 40V to drive the MOSFETs. Let's say we connect an inverted charger. Will the DSG pin still be on 40V or will it go to 0V?

Either way, if the DSG pin stays on 40V it will draw about 59mA with an inverted charger, which seems a lot. With DSG pin at 0V it will still draw around 6mA. Trying to invert the diode D16, almost no current will be drawn from the DSG pin, since voltage now drops over the 68kOhm resistor. (And of course turn off/on times of the MOSFETs change: What do we want? Fast turn off and slow turn on? Seems like we can't have both here.)

Please check the attached image of another TI design reference: [https://www.ti.com/lit/an/sluaa09a/sluaa09a.pdf](url)
The marked diode is inverted there.

[image]