LibreSolar/bms-c1 #42: Thermal Performance Tests

pasrom, 2023-10-09:
### Introduction:
Our aim was to conduct a thermal investigation of the BMS to determine how hot it gets at a current of 100 A or more.

### Test Setup:
The picture below illustrates the BMS setup during the testing phase (with heatsink). The tests were performed without actual battery, as described [here](https://github.com/LibreSolar/bms-c1/blob/main/testing/v0.3/README.md#11-test-setup).
[image]
[image]

### Used Devices:
- Power source: E/A Power Supply PSB 10750-120
- Sink: Deutronic xxx
- Temperature measurements: "TIM QVGA/HD" thermal imaging camera
- Linux & Windows Computer

### Boundary Conditions:
- Ambient Temperature: ~24°C

### Test Scenarios:
Three different scenarios were tested:
1. **Without Heatsink**:
   - Current: 100 A
   - Measured Temperature: ~91°C after 35 minutes
   - BMS Position: horizontally on the table
   - The thermal camera image below displays the temperature distribution upon concluding the test, highlighting the MOSFETs as the area of highest temperature.
   [image]

2. **With Heatsink (1)**:
   - Current: 100 A
   - Measured Temperature: ~49°C after 25 minutes
   - BMS Position: 5 cm above the table (floating, cooling fins facing down to the table)
   -  The thermal camera image below displays the temperature distribution upon concluding the test, with the PCB layer of the shunt being the hottest area.
   [image]

3. **With Heatsink (2)**:
   - Current: 122 A
   - Measured Temperature: ~66°C after 4 hours
   - BMS Position: identical to Measurement 2
   - The thermal camera image below displays the temperature distribution upon concluding the test, with the PCB layer of the shunt being the hottest area.
   [image]
   
   - Below graph shows the current profile, temperature rise of the internal sensors and thermal camera temperature areas over time.
   [image]
### Conclusion
The BMS can comfortably manage over 100 A while maintaining a temperature within an acceptable range.
If feasible, the PCB layer should have a thickness greater than 70 µm to minimize the heating around the shunt.
Further work could involve identifying the maximum current that results in a temperature of around 80°C.

pasrom, 2024-10-22:
@martinjaeger I performed an 80 A test without the heatsink using the latest hardware.

## Test Setup:
Libre Solar BMS C1 v0.4.1 (03/2024)

[image]
<div style="display: flex; justify-content: space-between;">
  [image]
  [image]
</div>
Left: 80 A power source. Right: 80 A sink.

## Boundary Conditions:

Ambient Temperature: ~21°C

## Test Scenario 80 A without heatsink

- Current: 80 A
- Measured Temperature: ~58°C after 60 minutes
- BMS Position: horizontally on the table
- The thermal camera image below displays the temperature distribution upon concluding the test, highlighting the MOSFETs as the area of highest temperature.

[image]

Below graph shows the current profile, temperature rise of the internal sensors and thermal camera temperature areas over time.
[image]

martinjaeger, 2024-10-24:
Nice, thanks for posting the test results!

pasrom, 2024-10-24:
I added the 100 A test without a heatsink to the post. I believe 80 A without a heatsink is acceptable at temperatures below 60 °C, but 100 A without one is not recommended because it reaches nearly 90 °C.