LibreSolar/bms-c1 #43: Overvoltage and Undervoltage Testing Issue with BMS_C1 Board

Ananyaaynana, 2023-11-02:
Issue Description:

We are currently in the process of testing the BMS C1 board, specifically by subjecting it to overvoltage and undervoltage tests. The expected behavior during an overvoltage or undervoltage condition is that the voltage across the PACK+ and PACK- terminals should be 0V. To monitor this, we have employed an oscilloscope to probe TP4 (charge pin of BQ76952) and TP6 (discharge pin of BQ76952) on the board.

However, we have encountered an issue with the behavior of TP6 during undervoltage conditions. According to our expectations, TP6 should remain at 0V during an undervoltage condition, but we have observed that the voltage at TP6 is fluctuating instead of staying at a stable 0V. Simultaneously, TP4 exhibits the anticipated high voltage.

Steps to Reproduce:

    1) Set up the BMS C1 board for undervoltage testing.
    2) Monitor the voltage at TP6 using an oscilloscope.
    3) Observe the fluctuating voltage at TP6 during the undervoltage condition.

Expected Behavior:

During an undervoltage condition, the voltage at TP6 should remain stable at 0V.

Actual Behavior:

The voltage at TP6 fluctuates instead of staying at a stable 0V.

Additional Information:

We have attached a snippet of the oscilloscope output when the probe is positioned at TP6 to provide a visual representation of the issue.

Attachment:

https://github.com/LibreSolar/bms-c1/assets/131510581/0ddd5d20-eb06-470d-b959-e04d02ece42b

Environment:

    Board: BMC C1. 
    Oscilloscope model: KEYSIGHT EDUX1052A.
    Testing conditions: Positive terminal of oscilloscope placed at TP6 and negative terminal connected to BAT-.

Impact:

This issue may affect the accuracy and reliability of the BMS C1 board's undervoltage protection, potentially leading to unintended consequences in real-world applications.

We kindly request assistance in investigating and resolving this issue to ensure the board's proper functionality under undervoltage conditions.

Labels: #testing #undervoltage

martinjaeger, 2023-11-02:
Thanks for posting this issue.

I could not reproduce the behavior with [latest main](https://github.com/LibreSolar/bms-firmware/tree/55c2ea2cf5a2658ad180f9c9eff17e8308848b03) firmware on a BMS C1 v0.3.3 board with 4s cells. Once the BMS goes into undervoltage state the TP6 stays at around 0V all the time and only goes up again after the battery voltage rises above the reconnect threshold.

Which firmware version did you use?

martinjaeger, 2023-11-07:
Did you also try with the most recent firmware?

Also: If it appears only on one board, it may actually be a hardware issue.

Ananyaaynana, 2023-11-10:
Follow-up: Live Data Not Publishing, and Board Shutdown Issue with Latest Firmware

Issue:
I tried updating to the latest firmware version "v23.1-1-g55c2ea2-dirty," but encountered two issues:

    Live data is not being published on Cutecom.
    The board shuts down when attempting to flash the code and can only be turned on using the S2 button.

Steps Taken:

To address the live data issue, I created an overlay file (as in https://libre.solar/bms-firmware/src/dev/customization.html) in the app folder with the following code:
{
    chosen {
        thingset,serial = &usb_serial;
    };
}

I then built and flashed the code using the "west flash" command. When attempting to monitor the board using the "west espressif monitor" command, I encountered the following output:
[image]

Additionally, when I tried to flash the code again, red and green leds turn off but there is voltage at the bat+, bat-, pack+ and pack-. When the leds are off the  "west espressif monitor"command  gives this:
[image]

On exiting the espressif monitor not able to flash code again. On holding down the S2 button the leds turn back on after blinking for a few seconds:
[image]

I would appreciate any guidance on resolving these issues. Specifically, I need help with:

    1) Diagnosing and resolving the issue with live data not being published on Cutecom.
    2) Understanding why the board is shutting down during code flashing and how to prevent this from happening.

Additional Information:

    Board Model:  Libre Solar BMS C1 (v0.3)
    Firmware Version: v23.1-1-g55c2ea2-dirty.

martinjaeger, 2023-11-11:
If an issue is unrelated to the topic of this issue, please open a new one.

Live data publication on the USB serial is now disabled by default. As mentioned in the [release notes](https://github.com/LibreSolar/bms-firmware/releases/tag/v23.1), the Zephyr shell is now exposed to the USB serial. In order to enable publication messages on the shell, enable it [here](https://github.com/LibreSolar/bms-firmware/blob/cb89c75b74e85a036b7c6243424e31c68b3b3018/app/boards/bms_c1.conf#L24).

Regarding your problem with flashing: Did you keep the serial monitoring on while you tried to flash (espressif monitor or CuteCom)? That's not possible, as flashing uses the same serial interface.

Ananyaaynana, 2023-11-11:
Okay I will raise a new issue.