LibreSolar/bms-c1 #44: BMS non-responsive after working for a few minutes.

Morynmoh, 2023-11-27:
Test on BMS carried out with mild success. 

BMS powered through cell connectors - 3 cells used. We were even able to connect to it via BLE using the Thingset App.

However, when we connected the BAT - and then BAT +, the unit became non-responsive. The status LED lights are all off and the unit cannot be found on BLE. Tried the reset S2 and on/off S1 buttons in vain.

martinjaeger, 2023-11-27:
Thanks for the report.

Did you also short the unused cell channels? Basically all cell connections from Cell3+ onwards?

Also can you see if there is something printed on the virtual COM port if you connect to the USB?

Does it restart if you disconnect BAT+ and BAT- again?

Morynmoh, 2023-11-27:
Thanks for the quick response.

We initially connected 3 cells without shorting the unused cell channels (which worked shortly). Later after the BMS went unresponsive, we connected all 16 cells (it didn't work). 

It doesn't work when we disconnect BAT + and BAT -. 

I will explore the virtual COM port option.

Morynmoh, 2023-11-27:
We have 16 cells but we trialed with 3cells to test the BMS. We only added upto 16 to see if that will turn on the BMS.

The battery voltage was 9.6V for the 3 cell.  [image]

Morynmoh, 2023-11-29:
Hi @martinjaeger - The USB virtual is nonresponsive . The Serial com port issue. 
[image]