LibreSolar/bms-c1 #47: Clarification needed on copper thickness of inner PCB layers

Selvakumar58, 2023-12-22:
In the PCB notes, we see the following text:

[image]

However, in the Board setup, the details are as follows:

[image]

Here the units are in mils, but when converted to um, we get:

105 um outer layers
35 um inner layers

Which is the right one?

Regards,
Selva.

pasrom, 2023-12-22:
Hello @Selvakumar58,

Could you confirm if you are using the latest commit? I noticed that in [PR #46 of the LibreSolar/bms-c1 repository](https://github.com/LibreSolar/bms-c1/pull/46), there doesn’t seem to be any difference. If you are not on the latest commit, it might be worth updating to see if that addresses your issue.

Selvakumar58, 2023-12-26:
Hello @pasrom ,

We are using V0.3.3 (Commit no: b7895e0fe39fa203ef21a05e899bce7c4bfe8ab3) to build a few prototypes before transitioning to the latest commit. 

Any thoughts on the copper thicknesses for this commit?

Regards,
Selvakumar.

pasrom, 2023-12-26:
From what I understand, @martinjaeger manufactures his boards with a 70 μm thickness for all four layers. It would be great if @martinjaeger could confirm this.

martinjaeger, 2023-12-27:
Correct. We manufactured the first batch with 70u for inner and outer layer because the originally intended 105u were not suitble for the small pin pitch of the bq76952.

Both myself an @pasrom have tested the boards and 70u turns out to be sufficient for 100A with a heat sink.