LibreSolar/bms-c1 #51: In file bq769x2_ In emul. c, The size defined by subcmd_mem seems to exceed the actual required size

yangminglong, 2024-01-15:
In file bq769x2_ In emul. c, The size defined by subcmd_mem seems to exceed the actual required size
`/*
 * Memory layout of bq769x2 for direct commands and subcommands
 *
 * Used subcommand address space starts with 0x9180 and ends below 0x9400
 */
#define BQ_DIRECT_MEM_SIZE (0x80)
#define BQ_SUBCMD_MEM_SIZE (0x9400)

struct bq769x0_emul_data
{
    /* Memory of bq769x2 for direct commands */
    uint8_t direct_mem[BQ_DIRECT_MEM_SIZE];
    /* Memory of bq769x2 for subcommands / data */
    uint8_t subcmd_mem[BQ_SUBCMD_MEM_SIZE];
    uint32_t cur_reg;
};`

martinjaeger, 2024-01-15:
Yes, it consumes more memory than required. The reason is to simplify the calculation such that we can directly use the bq register address for the position in the array.

The unit tests are normally run on a desktop computer where the amount of wasted memory doesn't matter. Did you run into any issues?

yangminglong, 2024-01-15:
Thank you for your reply. I haven't encountered any problems. This issue can be closed now

closed 2024-01-15 by martinjaeger