LibreSolar/bms-c1 #55: Fix NPTH clearance in v0.4

devbotas, 2024-03-06:
aisler.net complains about this:
[image]

And indeed, one of those USB connector holes got electroplated, shorting 3V3 to GND. 

I (hopefully) fixed my PCBs by drilling a slightly larger hole.

martinjaeger, 2024-03-06:
Thanks a lot for reporting this issue.

I had ordered one PCB for validating the design, which worked fine. However, the NPTH of the USB connector looks indeed wrong. Also the KiCAD DRC didn't complain about this. That's very annoying, as I have already ordered a batch of assembled boards.

Will fix the issue and keep you updated.

martinjaeger, 2024-03-07:
The good news is: I could stop the assembly just before placement of the parts. I will get the PCBs for double-checking. Most probably we will have to re-order them.

@devbotas Your bug report came just in time. Thanks again! If you like, I can send you a fully assembled v0.3.3 board for free, so that you can immediately start working with the BMS. Please send me an email (martin at libre solar) with contact details.

devbotas, 2024-03-07:
Great to hear! Although your PCBs may still be OK.  The electrical test of my first batch was fine (or maybe they didn't do it, I don't know), but it was lost in shipping. It is the second batch that resulted in faulty tests.

martinjaeger, 2024-03-17:
@devbotas FYI: I tested the insulation resistance of approx. 20 of the v0.4.0 boards with a Fluke 1507 and got >11 GOhm (max. of measurement range) for all of them for a test voltage of 1000 V. Looks like these PCBs were fine. However, the clearance should still be fixed, of course.

Please see the PR linked above for the fix.

devbotas, 2024-03-17:
So you've got lucky, nice 👍

commit 7545e526 (2024-04-06): Increase clearance around NPTH (v0.4.1)

The NPTHs of the USB connector are very close to the pads, such that a
clearance of 0.25 mm as required by some manufacturers was not possible.
The clearance is now 0.2 mm.

Fixes #55
PR #58 (2024-04-06): Increase clearance around NPTH (v0.4.1)
The NPTHs of the USB connector are very close to the pads, such that a clearance of 0.25 mm as required by some manufacturers was not possible. The clearance is now 0.2 mm.

Fixes #55
closed 2024-04-06 by martinjaeger