LibreSolar/bms-c1 #63: Support for daisy chaining (or otherwise connecting) multiple LibreSolar BMS for stackable energy storage packs

rexlee2, 2024-07-31:
To connect multiple BMS in daisy chain manner (e.g. to implement stackable energy storage module) it seems that using CAN 2.0 bus would be most appropriate.

Can 11 bit DIP switches be added to assign 11-bit CAN ID supported by CAN 2.0A standard? Given the price of batteries, I believe LibraSolar should assume that most residential and commercial applications will require an easy method of scaling LibreSolar BMS without having to load unique BMS software to every single BMS unit within energy storage pack.

Appreciate your feedback and consideration!

martinjaeger, 2024-07-31:
The ThingSet protocol (used by default by the Libre Solar BMS) does already support automatic node address assignment, see [here](https://thingset.io/spec/v0.6/transports/can.html#network-management) for the details.

Daisy-chaining is more an electrical question. You need to make sure that you only connect two BMSs in parallel which have the same SOC and capacity. So you'd need to start communication before you switch on the MOSFETs. This is currently not implemented, but could be done on top of ThingSet.

Connecting multiple batteries in series is currently not supported, as it would require MOSFETs rated with the resulting overall pack voltage, which is not the case.

rexlee2, 2024-08-01:
Thanks for the detailed explanation Martin! This helps a lot.
I understand that the BMS units need to communicate and ensure that the SOCs and other parameters are aligned. Any thought or comment on how similar SOC or voltage levels need to be before turning on the MOSFETs? I wonder if there is a SOC mismatch that would be considered unsafe for parallel connection.