LibreSolar/bms-c1 #65: Power Terminal blocks

fl4p, 2024-11-30:
Hi,

I am curious, why do you use SMD power terminals for the battery connection?
These appear to have low availability and a bit pricy.
I think THT has better mechanical stability and can directly connect to all copper layers.

This would be a cheap alternatives:
[M6 Nut ](https://www.digikey.com/en/products/detail/w%C3%BCrth-elektronik/S93515/10123864)
[M5 Screw](https://www.digikey.com/en/products/detail/w%C3%BCrth-elektronik/S93514/10123859)

fotherja, 2024-12-05:
Yes, I agree. I feel THT terminal blocks would carry the currents to all 4 copper layers better and be more robust overall. This is a nice part: [16 pin M5 180A rated](https://www.mouser.co.uk/ProductDetail/Wurth-Elektronik/7461383?qs=lykWx4dhCCFZ0JBsRfYENw%3D%3D)

fotherja, 2024-12-05:
Yes, I agree. I feel a THT terminal block would better carry the current to all 4 copper layers and be more robust. Here a nice [M5 Nut 180 A solution](https://www.mouser.co.uk/ProductDetail/Wurth-Elektronik/7461383?qs=lykWx4dhCCFZ0JBsRfYENw%3D%3D)

martinjaeger, 2024-12-11:
Thanks for the suggestions, @fl4p and @fotherja.

We considered THT parts, but could not find any suitable parts for the required amount of current which are available at the usual distributors.

As far as I know, the parts you linked are not "normal" THT parts, but press-fit parts. We actually used press-fit parts in previous versions of this BMS, but finally decided against them because they require special PCB treatment which is more expensive and not available at all PCB manufacturers.

One more advantage of the SMD terminals is that the bottom of the PCB remains completely flat, so that it can be easily mounted to a heat sink. As soon as you have press-fit terminals or THT terminals you will need milling / cutouts in the heat sink.