LibreSolar/bms-c1 #68: Voltage spike damaging LMR during power-on with power supply and electronic load

pasrom, 2024-12-09:
During recent testing #67 of the short-circuit shutdown via MOSFETs on the BMS C1, I noticed an issue where turning on the lab power supply with an attached electronic load occasionally causes a voltage spike that damages the LMR38010. After replacing it, the BMS C1 functions normally again. The following figure shows a successful shutdown where the power supply reached the configured voltage limit and turns off. Unfortunately, I couldn’t capture a failure case.

[image]

We suspect this issue might be caused by the interaction between the electronic load and the power supply.
Has anyone else experienced something similar?

martinjaeger, 2025-01-28:
Thanks @pasrom for reporting the issue. Only found time today to try to reproduce it.

I used a bi-directional power supply from EA for the BAT+/- side (set to 12V, +-10A) and a "normal" one connected to PACK+/- (set to 14.4V, 10A). I started with only 12V, as I had a resistor network ready for that voltage to simulate the single cell voltages.

I can see a voltage spike higher than the voltage provided by the power supply upon connection of the power supply to PACK+/-:

[image]

When switching the CHG MOSFET on/off via ThingSet `thingset =Input {\"wChgEnable\":true}`, the spike is much lower, most probably due to pre-charging.

[image]

I'm guessing that the voltage spike is coming from the inductivity of the wires. My wires were approx. 2m long (both on BAT and PACK side).

Normally, D12 (SMCJ70A) should cut off voltage spikes below the max. voltage of the LMR3810 (85V). So I'm not exactly sure why it got damaged.

Did you find out anything new in the meantime, @pasrom? And how did you measure the current in your experiment?

pasrom, 2025-01-29:
Thanks a lot for trying it out.

To measure the current, I used a Teledyne LeCroy CP150-6M.

Ah, that explains why I didn’t see the pre-charging, the MOSFETs are already opening automatically. I should test it as intended: starting the load only when the MOSFETs are disabled (thingset = Input {"wChgEnable": false}). Thanks for pointing me in this direction.

What do you think, could the maximum breakdown voltage (V<sub>BR</sub>) of 86 V interfere with the LMR38010, since it exceeds its absolute maximum rating of 85 V?
I could solder an SMCJ64A instead of the SMCJ70A and test it again. I already have a BMS where I desoldered the SMCJ70A after it was damaged by accidental overvoltage.

[image]

martinjaeger, 2025-02-03:
> I could solder an SMCJ64A instead of the SMCJ70A and test it again. I already have a BMS where I desoldered the SMCJ70A after it was damaged by accidental overvoltage.

I doubt it would make a huge difference. If we have peak currents after switching off far beyond 10A, the TVS diode won't be able to handle properly anyway. I'll try to come up with an LT spice simulation to how much energy the spikes would have for typical cable lengths and large currents.

Jan-vi, 2025-02-10:
Did you consider intrinsic inductors of power line and cells? Assume 1uH before Mosfet and 0,5uH behind Mosfet. To test Mosfet current breaking  capability, a maximum load inductor needs to be specified.  Big inductivity gives time to switch off early. On the other hand, current goes on when Mosfet switches off. For several microseconds, there is full current and battery voltage over drain/source. This requires to have a TVS for both drain and source of Mosfet with short GND connection. At switch-off moment, my battery voltage goes up to about 85 Volt.  For a few microseconds, the Mosfets SOA is challenged with several kilowatts to survive.

martinjaeger, 2025-04-30:
@pasrom Do you remember some more details of the setup where the overvoltage spike happened, which destroyed the LMR? We are building up a more detailed simulation model.

The following points would be interesting:
- Wire length and cross-section
- Cell capacity and internal resistance
- Pack voltage (or number of cells)

pasrom, 2025-05-09:
@martinjaeger thank you for doing a more detailed simulation model.
- **Wire length and cross‑section**
Wire length (one path): ~ 2 m
Cross‑section: 25 mm²
- **Cell capacity and internal resistance**
I used two bidirectional power supplies when this situation occurred:
  - Source: [PSB 1075-120](https://elektroautomatik.com/shop/de/produkte/programmierbare-dc-laborstromversorgungen/dc-bidirektionale-laborstromversorgungen/dc-bidirektionale-laborstromversorgungen-psb-10000-4u/936/bidirektionales-labornetzgeraet)
  - Sink: [IT-M3900C](https://www.itechate.com/en/product/dc-power-supply/IT-M3900C.html)
- **Pack voltage**
Simulated pack voltage: 60 - 65 V

knakwaboah, 2026-01-03:
> Thanks a lot for trying it out.
> 
> To measure the current, I used a Teledyne LeCroy CP150-6M.
> 
> Ah, that explains why I didn’t see the pre-charging, the MOSFETs are already opening automatically. I should test it as intended: starting the load only when the MOSFETs are disabled (`thingset = Input {"wChgEnable": false}`). Thanks for pointing me in this direction.
> 
> What do you think, could the maximum breakdown voltage (VBR) of 86 V interfere with the LMR38010, since it exceeds its absolute maximum rating of 85 V? I could solder an SMCJ64A instead of the SMCJ70A and test it again. I already have a BMS where I desoldered the SMCJ70A after it was damaged by accidental overvoltage.
> 
> <img alt="Image" width="701" src="https://private-user-images.githubusercontent.com/32713472/407876365-bc6f361e-e402-4a33-9ea9-8245adc39580.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NjczOTg4MTAsIm5iZiI6MTc2NzM5ODUxMCwicGF0aCI6Ii8zMjcxMzQ3Mi80MDc4NzYzNjUtYmM2ZjM2MWUtZTQwMi00YTMzLTllYTktODI0NWFkYzM5NTgwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNjAxMDMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjYwMTAzVDAwMDE1MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTc5YzkxMDE0ZGMwNDhjY2I1ZjJjYjliZWYwNmNjMzQ3YjQ4N2YzM2Y3OTY4YzI5OGQ3YTE4NWFhZjAyNmMxZjMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.0_dOEAWjzwzcSSKojzO3w7DLYelx872ND00swhgpUmI">

 Based on the datasheet, the SMCJ70A TVS clamps at 113V, but the LMR38010/BQ76952 have an 85V absolute max. During a surge, the voltage will overshoot above 85V before clamping thereby damaging the ICs even though the TVS is working.
Should we use a TVS with a lower clamping voltage (like an SMCJ64CA) to ensure the clamped voltage never exceeds 85V?
If this does not work, then we can look into  the addition of filters or controlling the starting current for the FETs so that voltage build up due to the inductance in the setup can be controlled.Please let me know what you think. Also any outcome of the simulation energy generated due to the spikes since that can give us some guidance into the amount of current to expect during clamping.